physical-computation/Warp-firmware #38: Issue compiling latest Warp on Ubuntu 18.10 virtual machine. Error: region  `m_text' overflowed by 1872 bytes

Meech0, 2019-01-10:
When attempting to compile Warp on my Ubuntu 18.10 virtual machine I get the print out found in compile.txt.  

The output from running gcc -v can be found in gcc.txt. 

[compile.txt](https://github.com/physical-computation/Warp-firmware/files/2747065/compile.txt)
[gcc.txt](https://github.com/physical-computation/Warp-firmware/files/2747066/gcc.txt)

Siegfriedchao, 2019-01-10:
```
douglas@douglas-VirtualBox:~$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 8.2.0-7ubuntu1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 8.2.0 (Ubuntu 8.2.0-7ubuntu1) 
```

Siegfriedchao, 2019-01-10:
```
douglas@douglas-VirtualBox:~$ cd Warp-firmware
douglas@douglas-VirtualBox:~/Warp-firmware$ cd build/
douglas@douglas-VirtualBox:~/Warp-firmware/build$ cd ksdk1.1/
douglas@douglas-VirtualBox:~/Warp-firmware/build/ksdk1.1$ ./build.sh
-- TOOLCHAIN_DIR: /usr
CMake Deprecation Warning at /usr/share/cmake-3.12/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/douglas/Warp-firmware/build/ksdk1.1/work/tools/cmake_toolchain_files/armgcc.cmake:38 (CMAKE_FORCE_C_COMPILER)
  /usr/share/cmake-3.12/Modules/CMakeDetermineSystem.cmake:91 (include)

CMake Deprecation Warning at /usr/share/cmake-3.12/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/douglas/Warp-firmware/build/ksdk1.1/work/tools/cmake_toolchain_files/armgcc.cmake:39 (CMAKE_FORCE_CXX_COMPILER)
  /usr/share/cmake-3.12/Modules/CMakeDetermineSystem.cmake:91 (include)

-- BUILD_TYPE: Release
-- TOOLCHAIN_DIR: /usr
CMake Deprecation Warning at /usr/share/cmake-3.12/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/douglas/Warp-firmware/build/ksdk1.1/work/tools/cmake_toolchain_files/armgcc.cmake:38 (CMAKE_FORCE_C_COMPILER)
  CMakeFiles/3.12.1/CMakeSystem.cmake:6 (include)

CMake Deprecation Warning at /usr/share/cmake-3.12/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/douglas/Warp-firmware/build/ksdk1.1/work/tools/cmake_toolchain_files/armgcc.cmake:39 (CMAKE_FORCE_CXX_COMPILER)
  CMakeFiles/3.12.1/CMakeSystem.cmake:6 (include)

-- BUILD_TYPE: Release
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/douglas/Warp-firmware/build/ksdk1.1/work/lib/ksdk_platform_lib/armgcc/KL03Z4
Scanning dependencies of target KsdkPlatformLib
[  1%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/osa/src/fsl_os_abstraction_bm.c.obj
[  5%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/system/src/clock/fsl_clock_manager_common.c.obj
[  3%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/system/src/clock/fsl_clock_manager.c.obj
[  7%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/system/src/clock/MKL03Z4/fsl_clock_MKL03Z4.c.obj
[  9%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/system/src/interrupt/fsl_interrupt_manager.c.obj
[ 11%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/system/src/hwtimer/fsl_hwtimer.c.obj
[ 13%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/system/src/hwtimer/fsl_hwtimer_systick.c.obj
[ 15%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/system/src/power/fsl_power_manager.c.obj
[ 17%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/hal/src/adc16/fsl_adc16_hal.c.obj
[ 19%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/adc16/fsl_adc16_driver.c.obj
[ 21%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/hal/src/cmp/fsl_cmp_hal.c.obj
[ 23%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/adc16/fsl_adc16_common.c.obj
[ 25%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/cmp/fsl_cmp_driver.c.obj
[ 26%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/cmp/fsl_cmp_common.c.obj
[ 28%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/hal/src/cop/fsl_cop_hal.c.obj
[ 30%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/cop/fsl_cop_driver.c.obj
[ 32%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/cop/fsl_cop_common.c.obj
[ 34%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/hal/src/gpio/fsl_gpio_hal.c.obj
[ 36%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/gpio/fsl_gpio_common.c.obj
[ 38%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/douglas/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/gpio/fsl_gpio_driver.c.obj
[... cut ...]
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `m_text' overflowed by 1872 bytes
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `m_data' overflowed by 168 bytes
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/Warp.dir/build.make:467: release/Warp.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/Warp.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

Now, run

	/Applications/SEGGER/JLink/JLinkExe -device MKL03Z32XXX4 -if SWD -speed 4000 -CommanderScript ../../tools/scripts/jlink.commands
```

phillipstanleymarbell, 2019-08-30:
This is a compiler issue. Moving to WontFix.

Ocoras, 2019-11-03:
I am getting a similar error when building Warp-firmware on Arch Linux using arm-none-eabi-gcc version 9.2.0:  `Warp.elf section '.stack' will not fit in region 'm_data'`. If this remains a compiler error yet I am using the latest compiler, is there anything that can be done to solve this? There are no bug reports on the Arch package for this, and I cannot find the relevant bug on the gcc bug tracker.

Script output:
```bash
[david@derobe ksdk1.1]$ ./build.sh 
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- TOOLCHAIN_DIR: /usr
CMake Deprecation Warning at /usr/share/cmake-3.15/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/david/Warp-firmware/build/ksdk1.1/work/tools/cmake_toolchain_files/armgcc.cmake:38 (CMAKE_FORCE_C_COMPILER)
  /usr/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:90 (include)

CMake Deprecation Warning at /usr/share/cmake-3.15/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/david/Warp-firmware/build/ksdk1.1/work/tools/cmake_toolchain_files/armgcc.cmake:39 (CMAKE_FORCE_CXX_COMPILER)
  /usr/share/cmake-3.15/Modules/CMakeDetermineSystem.cmake:90 (include)

-- BUILD_TYPE: Release
-- TOOLCHAIN_DIR: /usr
CMake Deprecation Warning at /usr/share/cmake-3.15/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/david/Warp-firmware/build/ksdk1.1/work/tools/cmake_toolchain_files/armgcc.cmake:38 (CMAKE_FORCE_C_COMPILER)
  CMakeFiles/3.15.5/CMakeSystem.cmake:6 (include)

CMake Deprecation Warning at /usr/share/cmake-3.15/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/david/Warp-firmware/build/ksdk1.1/work/tools/cmake_toolchain_files/armgcc.cmake:39 (CMAKE_FORCE_CXX_COMPILER)
  CMakeFiles/3.15.5/CMakeSystem.cmake:6 (include)

-- BUILD_TYPE: Release
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/david/Warp-firmware/build/ksdk1.1/work/lib/ksdk_platform_lib/armgcc/KL03Z4
Scanning dependencies of target KsdkPlatformLib
[  5%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/system/src/clock/fsl_clock_manager.c.obj
[  5%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/osa/src/fsl_os_abstraction_bm.c.obj
[  5%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/system/src/clock/fsl_clock_manager_common.c.obj
[  7%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/system/src/clock/MKL03Z4/fsl_clock_MKL03Z4.c.obj
[  9%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/system/src/interrupt/fsl_interrupt_manager.c.obj
[ 11%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/system/src/hwtimer/fsl_hwtimer.c.obj
[ 13%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/system/src/hwtimer/fsl_hwtimer_systick.c.obj
[ 15%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/system/src/power/fsl_power_manager.c.obj
[ 17%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/hal/src/adc16/fsl_adc16_hal.c.obj
[ 19%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/adc16/fsl_adc16_driver.c.obj
[ 21%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/adc16/fsl_adc16_common.c.obj
[ 23%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/hal/src/cmp/fsl_cmp_hal.c.obj
[ 25%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/cmp/fsl_cmp_driver.c.obj
[ 26%] Building C object CMakeFiles/KsdkPlatformLib.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/drivers/src/cmp/fsl_cmp_common.c.obj
[... 106 lines cut ...]
[ 85%] Building C object CMakeFiles/Warp.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/startup/MKL03Z4/system_MKL03Z4.c.obj
[ 90%] Building C object CMakeFiles/Warp.dir/home/david/Warp-firmware/build/ksdk1.1/work/platform/startup/startup.c.obj
[... cut ...] reads --disable-tls --with-gnu-as --with-gnu-ld --with-system-zlib --with-newlib --with-headers=/usr/arm-none-eabi/include --with-python-dir=share/gcc-arm-none-eabi --with-gmp --with-mpfr --with-mpc --with-isl --with-libelf --enable-gnu-indirect-function --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='Arch Repository' --with-bugurl=https://bugs.archlinux.org/ --with-multilib-list=rmprofile
Thread model: single
gcc version 9.2.0 (Arch Repository) 
```