physical-computation/Warp-firmware #39: In 'z' command hex sensor data is cast to int and printed. It needs to be dealt with properly to preserve sign

Meech0, 2019-01-24:
Read through data sheets and determine how to interpret the hex values from each sensor to preserve sign and produce c code that maps the raw hex value to an unsigned integer from 0 to 2^n. For example for at +- 2g value of 16 bits -2g should be mapped to 0 and 2g should be mapped to 65536. this would lead to 0 g being mapped to 65536/2

Siegfriedchao, 2019-01-26:
Side note: The MMA8451Q sensor has 8-bit and 14-bit outputs. The output is in 2's complement format.
You'll be able to retrieve the signed data for the negative values (pseudocode)
```
if outputdata >= 8192 then do
        outputdata = outputdata - 16384
end if
```

Meech0, 2019-01-27:
Thanks Youchao I have implemented that, you saved me a lot of time :)

phillipstanleymarbell, 2019-08-30:
Cleaned up in #41. Moving to Verify.

closed 2020-01-31 by phillipstanleymarbell