physical-computation/Warp-firmware #70: Start of measurement number

richardhenryhopper, 2020-01-27:
Measurement number starts higher than 0 (e.g. >800).

richardhenryhopper, 2020-01-29:
Tried but with no success:

**Making readingCount a global variable** 
- Uncommenting sensor #includes.
- Changing the variable type to int, uint16_t

**Tested SEGGER_RTT_printf**
Prints OK when setting readingCount set an arbitrary value e.g.:

readingCount = 4294967295;
SEGGER_RTT_printf(0, "%u, %d, %d,", readingCount, RTC->TSR, RTC->TPR);
Output: 4294967295;

Set readingCount after printf statement:
SEGGER_RTT_printf(0, "%u, %d, %d,", readingCount, RTC->TSR, RTC->TPR);
readingCount = 0;
Output: 1 (incremented as expected)

richardhenryhopper, 2020-01-30:
Issue only occurs when running JLinkRTTLogger (not JLinkRTTClient).

phillipstanleymarbell, 2020-01-31:
This is not a bug / resolved. Closing.

closed 2020-01-31 by phillipstanleymarbell