physical-computation/Warp-firmware #73: Sensor output values are wrong

vimuth, 2020-02-03:
**Describe the bug**
Sensor output values from the "z" command are wrong. 
Some sensors always output 0. For example the HDC1000 sensor always outputs 0x00 in registers 0x00 and 0x01. Same output when looking specifically at that sensor using the "j" command.

**To reproduce**
1. Run ./build.sh in build/ksdk1.1/
2. Then run JLinkExe -device MKL03Z32XXX4 -if SWD -speed 100000 -CommanderScript <commands file> and JLinkRTTClient
3. Running the following set of commands to dump all sensor values
b 0300
r
g 3000
n
z

**Expected behavior**
I got the following output
BME680 Calibration Data: 0x40, 0xFB, 0x66, 0x03, 0xFF, 0x95, 0x8C, 0x59, 0xD7, 0x58, 0xFF, 0x90, 0x1E, 0x81, 0xFF, 0x22, 0x1E, 0x00, 0x00, 0x48, 0xF9, 0x60, 0xF4, 0x1E, 0x01, 0x3F, 0x94, 0x2E, 0x00, 0x2D, 0x14, 0x78, 0x9C, 0x53, 0x65, 0xF4, 0xDD, 0xE0, 0x12, 0x23, 0x00

Measurement number, RTC->TSR, RTC->TPR, AMG8834 0, AMG8834 1, AMG8834 2, AMG8834 3, AMG8834 4, AMG8834 5, AMG8834 6, AMG8834 7, AMG8834 8, AMG8834 9, AMG8834 10, AMG8834 11, AMG8834 12, AMG8834 13, AMG8834 14, AMG8834 15, AMG8834 16, AMG8834 17, AMG8834 18, AMG8834 19, AMG8834 20, AMG8834 21, AMG8834 22, AMG8834 23, AMG8834 24, AMG8834 25, AMG8834 26, AMG8834 27, AMG8834 28, AMG8834 29, AMG8834 30, AMG8834 31, AMG8834 32, AMG8834 33, AMG8834 34, AMG8834 35, AMG8834 36, AMG8834 37, AMG8834 38, AMG8834 39, AMG8834 40, AMG8834 41, AMG8834 42, AMG8834 43, AMG8834 44, AMG8834 45, AMG8834 46, AMG8834 47, AMG8834 48, AMG8834 49, AMG8834 50, AMG8834 51, AMG8834 52, AMG8834 53, AMG8834 54, AMG8834 55, AMG8834 56, AMG8834 57, AMG8834 58, AMG8834 59, AMG8834 60, AMG8834 61, AMG8834 62, AMG8834 63, AMG8834 Temp, MMA8451 x, MMA8451 y, MMA8451 z, MAG3110 x, MAG3110 y, MAG3110 z, MAG3110 Temp, L3GD20H x, L3GD20H y, L3GD20H z, L3GD20H Temp, BME680 Press, BME680 Temp, BME680 Hum, BMX055acc x, BMX055acc y, BMX055acc z, BMX055acc Temp, BMX055mag x, BMX055mag y, BMX055mag z, BMX055mag RHALL, BMX055gyro x, BMX055gyro y, BMX055gyro z, CCS811 ECO2, CCS811 TVOC, CCS811 RAW ADC value, HDC1000 Temp, HDC1000 Hum, RTC->TSR, RTC->TPR, # Config Errors

0, 1451692821, 20372, 0, 9, 10, 14, 1, 0, 0, 0, 0, 0, 0, 1, 0, 0, 8, 0, 0, 0, 0, 0, 9, 8, 0, 0, 8, 0, 0, 0, 0, 0, 4, 0, 12, 12, 12, 13, 12, 12, 12, 10, 13, 14, 13, 14, 14, 14, 14, 13, 14, 9, 14, 9, 16, 10, 16, 7, -510, 0, 0, 2, 1, 2, 0, -508, 8, 132, 694, -3938, -2083, 1879, 1737, 12, -91, -444, -137, 7, 524288, 524288, 32768, -91, 174, 995, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 43, -40, ----, 1451692821, 21505, 6
1, 1451692821, 24753, 0, 12, 8, 6, 1, 0, 0, 0, 0, 0, 0, 1, 0, 0, 8, 0, 0, 0, 0, 0, 9, 8, 0, 0, 8, 0, 0, 0, 0, 0, 4, 0, 12, 12, 12, 13, 12, 12, 12, 10, 13, 14, 13, 14, 14, 14, 14, 13, 14, 9, 14, 9, 16, 10, 16, 7, -510, 0, 0, 2, 1, 2, 0, -508, 8, 162, 664, -3966, -2081, 1876, 1740, 12, -64, -375, -147, 7, 359136, 496256, 18242, -84, 167, 1001, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 92, ----, 0, 1451692821, 25883, 6
2, 1451692821, 29134, 0, 13, 12, 2, 0, 0, 0, 0, 0, 0, 0, 2, 0, 0, 8, 0, 0, 0, 0, 0, 9, 10, 0, 0, 10, 0, 0, 0, 0, 0, 4, 0, 10, 10, 11, 11, 11, 11, 10, 8, 11, 11, 11, 11, 12, 11, 12, 10, 10, 7, 11, 8, 12, 8, 11, 5, 5, 8, 8, 8, 8, 9, 7, 5, 8, 132, 688, -3930, -2082, 1874, 1751, 12, -52, -349, -146, 7, 359136, 496288, 18232, -70, 177, 997, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 92, -40, ----, 1451692821, 30267, 6

The sensor readings seems to be wrong and to be disagreeing with one another.
And I always see 4/6 Config Errors. Is this expected?

**Screenshots and hardware pictures**
[image]
[image]
I am using the latest version of the firmware 
[image]

**Host OS (please complete the following information):**
 - OS: Ubuntu 18.04.3
 - Browser: Chrome
 - Version (e.g., 22).
 - J-Link Commander V6.56a

**You local changes (please complete the following information):**
 - Output of `git diff`.
diff --git a/tools/scripts/jlink.commands b/tools/scripts/jlink.commands
index e0d59ab..d449d69 100644
--- a/tools/scripts/jlink.commands
+++ b/tools/scripts/jlink.commands
@@ -1,5 +1,5 @@
 exec EnableRemarks
 unlock kinetis
-loadfile <full-path-to-warp-firmware>/build/ksdk1.1/work/demos/Warp/armgcc/Warp/release/Warp.srec
+loadfile /home/vimuth/projects/warp/Warp-firmware/build/ksdk1.1/work/demos/Warp/armgcc/Warp/release/Warp.srec
 r
 go
\ No newline at end of file

 - Output of `git remote -v`.
origin	git@github.com:physical-computation/Warp-firmware.git (fetch)
origin	git@github.com:physical-computation/Warp-firmware.git (push)

**Additional context**
I have set the switches for "Power from USB and Sensors Enabled" as in - https://github.com/physical-computation/Warp-hardware/tree/master/revB
all switches in S3 is set to off

phillipstanleymarbell, 2020-02-04:
Might be a physically-faulty HDC1000; sensors break.

@vimuth : other than the HDC1000, are there any other specific sensor values that are not working on your system?

phillipstanleymarbell, 2020-05-17:
No response from @vimuth . Closing.

closed 2020-05-17 by phillipstanleymarbell