physical-computation/Warp-firmware #78: CSS811 Register 0x06

JamesTimothyMeech, 2020-04-06:
**Describe the bug**
We currently only read 2 hex bytes from register 0x06 on the CSS811 and print them with the z command. 
The datasheet suggests that there are four bytes (two for vref anf two for a negative temperature coefficient sensor) to be read and all of them are needed to calculate the temperature see page 20: https://cdn.sparkfun.com/assets/learn_tutorials/1/4/3/CCS811_Datasheet-DS000459.pdf

**To reproduce**
See only two hex bytes printed in: https://github.com/physical-computation/Warp-data-stream/tree/master/data

**Expected behavior**
All four bytes should be printed to allow temperature calculation

phillipstanleymarbell, 2020-04-07:
@JamesTimothyMeech : Assigning to you.

phillipstanleymarbell, 2020-05-17:
Fixed in #83 by @JamesTimothyMeech .

closed 2020-05-17 by phillipstanleymarbell