physical-computation/Warp-firmware #111: Unknown type name error `caddr_t` when using Arm GNU Toolchain 12.2

KomaGR, 2023-02-13:
```
arm-none-eabi-gcc (Arm GNU Toolchain 12.2 (Build arm-12.24)) 12.2.1 20221205
```

```
/Users/orestis/physical-computation/Warp-firmware/build/ksdk1.1/work/platform/utilities/src/fsl_misc_utilities.c:71:1: error: unknown type name 'caddr_t'
   71 | caddr_t
      | ^~~~~~~
/Users/orestis/physical-computation/Warp-firmware/build/ksdk1.1/work/platform/utilities/src/fsl_misc_utilities.c: In function '_sbrk':
/Users/orestis/physical-computation/Warp-firmware/build/ksdk1.1/work/platform/utilities/src/fsl_misc_utilities.c:87:15: error: 'caddr_t' undeclared (first use in this function)
   87 |       return (caddr_t) -1;
      |               ^~~~~~~
/Users/orestis/physical-computation/Warp-firmware/build/ksdk1.1/work/platform/utilities/src/fsl_misc_utilities.c:87:15: note: each undeclared identifier is reported only once for each function it appears in
/Users/orestis/physical-computation/Warp-firmware/build/ksdk1.1/work/platform/utilities/src/fsl_misc_utilities.c:92:19: error: expected ';' before 'prev_heap_end'
   92 |   return (caddr_t) prev_heap_end;
      |                   ^~~~~~~~~~~~~~
      |                   ;
/Users/orestis/physical-computation/Warp-firmware/build/ksdk1.1/work/platform/utilities/src/fsl_misc_utilities.c:77:17: warning: variable 'prev_heap_end' set but not used [-Wunused-but-set-variable]
   77 |   char *        prev_heap_end;
      |                 ^~~~~~~~~~~~~
/Users/orestis/physical-computation/Warp-firmware/build/ksdk1.1/work/platform/utilities/src/fsl_misc_utilities.c:93:1: warning: control reaches end of non-void function [-Wreturn-type]
   93 | }
      | ^
make[3]: *** [CMakeFiles/Warp.dir/Users/orestis/physical-computation/Warp-firmware/build/ksdk1.1/work/platform/utilities/src/fsl_misc_utilities.c.obj] Error 1
```

matiasilva, 2023-02-14:
I can confirm this error as well. Did you find a way around it?

matiasilva, 2023-02-14:
I've submitted #112 to fix this.

KomaGR, 2023-02-14:
> I can confirm this error as well. Did you find a way around it?

> I've submitted #112 to fix this.

Thanks for the PR. Seems to be related to a change in the two latest major versions of the ARM toolchain (11.x and 12.x). Compiling with 9.x or 10.x was a success.